Xinyuan-LilyGO/TTGO_TWatch_Library #161: Cannot download the bin by using ESP32 Download Tool

xiaomi2503, 2022-03-22:
My watch is 2020 v2
I try to download "T-Watch2020-V2.bin" by using ESP32 Download Tool.
But the tool only shows "SYNC waiting for power up".
I have tried to push the button once/5sec/10sec but no response.
After times is up, shown ESP32 chip sync error.
Anyone can give me suggestion.

LilyGO, 2022-03-22:
The watch integrates an automatic download circuit. If this happens all the time, there may be a problem with the automatic download circuit.

closed 2022-03-22 by LilyGO

xiaomi2503, 2022-03-22:
I just bought 2-watches. Both of them are the same which cannot be download the firmware by using ESP32 Download Tool.
They are QFN24 Version with CH9102F. Is there any difference with CP2104 version ?
Is there any method to RESET the CPU from the TOP of PCB of the watch ?
I cannot found the RESET pin (CHIP_PU) of ESP32  to make power up sync.