Xinyuan-LilyGO/TTGO_TWatch_Library #162: Failed to compile SimpleWatch in Arduino IDE

andyli, 2022-04-13:
TTGO_TWatch_Library commit:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/commit/83f59b5f92429307f62bb8ae63078b584771c7ec

Arduino version:
1.8.19

Boards manager installed:
esp32 2.0.3-RC1 from https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

Compilation errors:
WiFiEvent_t::SYSTEM_EVENT_* are not members of 'arduino_event_id_t'

Compilation log (verbose mode):
https://gist.githubusercontent.com/andyli/63b1354f5db9ede3f44acf066a1992b7/raw/b1810e00f9c422172ccea282301a8ebefbb232b6/gistfile1.txt

andyli, 2022-04-13:
It compiles and works after I downgraded esp32 from 2.0.3-RC1 to 1.0.6.

LilyGO, 2022-04-20:
In the higher version of the ARDUINO-ESP32 library, it needs to be modified here

closed 2022-04-20 by LilyGO