Xinyuan-LilyGO/TTGO_TWatch_Library #163: MacOS 12.3.1 Monterey LilyGo T-Watch 2021 does not appear as serial device

frankcohen, 2022-04-14:
Hi T-Watch fans, I am on a MacBook Pro running Mac OS X 12.3.1 Monterey. I install the CH341 serial driver for Mac from http://www.wch-ic.com/downloads/CH341SER_MAC_ZIP.html. I restart. I connect the T-Watch using a USB C cable and the T-U2T adaptor. I open a Terminal window and do not see the T-Watch:

```
frankcohen@FCohenMac2020 /dev % ls tty*
tty				ttyse
tty.BLTH			ttysf
tty.Bluetooth-Incoming-Port	ttyt0
tty.JBLCharge3			ttyt1
ttyp0				ttyt2
ttyp1				ttyt3
ttyp2				ttyt4
```
etc...

I then plug in an ESP-Prog board. This board has a USB micro connector. I see

```
frankcohen@FCohenMac2020 /dev % ls tty*
tty				ttysc
tty.BLTH			ttysd
tty.Bluetooth-Incoming-Port	ttyse
tty.JBLCharge3			ttysf
tty.usbserial-1442100		ttyt0
tty.usbserial-1442101		ttyt1
tty.usbserial-5			ttyt2
tty.usbserial-6			ttyt3
ttyp0				ttyt4
ttyp1				ttyt5
```

What am I doing wrong?

-Frank

LilyGO, 2022-04-20:
T-WATCH-2021 has no integrated serial port chip, it needs to be used with U2T.

closed 2022-04-20 by LilyGO

fcohen, 2022-04-20:
I am using it connected with the U2T connector.