Xinyuan-LilyGO/TTGO_TWatch_Library #168: T-Watch-2019 GPS-M8N bottom board I2C port assignment changed

kristoffee, 2022-06-05:
GPS-M8N bottom PCB board I2C port assignment changed between v1.1 and v1.2. This prevents additional I2C sensors on the expansion board working on the same default pins. 
I2C pins for v1.1 = 21, 22
I2C pins for v1.2 = 25, 26

Please provide example code to handle additional sensor via the grove connector on the expansion board. Currently, If I start the sensor with pins 25, 26, it works but then other devices won't function. If I leave it on default pins, the devices on the main board function but not the one connected externally to the GPS-M8N bottom board.

Attached images showing the difference between version 1.1 and 1.2

[image]

[image]

lewisxhe, 2022-06-14:
Wire is used as the communication handle of twatch by default. If you want to use 25 and 26 as I2C communication, you can use Wire1

```
const uint8_t sda = 25;
const uint8_t scl = 26;
Wire1.begin(sda,scl);

```

kristoffee, 2022-07-03:
Thanks! 
It looks like when LILYGO_WATCH_HAS_TOUCH is defined, CapacitiveTouch defined in TTGO.h also uses Wire1. 
I managed to get my external I2C sensor working by performing the below adjustments. 
1. Changed #define LILYGO_WATCH_2019_WITH_TOUCH to #define LILYGO_WATCH_2019_NO_TOUCH,
2. Changed Wire to Wire1 in my external sensor's library,
3. Use the below initialization in the setup() function. 
const uint8_t sda = 25;
const uint8_t scl = 26;
Wire1.begin(sda,scl);

Hope the above helps someone else!

kristoffee, 2022-07-03:
Do you know why the PINS changed from 21, 22 to 25, 26 on the new board version?