Xinyuan-LilyGO/TTGO_TWatch_Library #172: new version with ULP possible with I2C pins GPIO 1/2 or GPIO 2/3

dlarue, 2022-09-24:
I'm only getting half a day use out of the battery and looked into using deep sleep and the ULP processor but the T-Watch does not have the proper I2C pins connected for ULP to be used.  The TSH-Watch can get 3-4 days on a 300mA battery albeit using e-INK display. See this issue post from the developer of TSH-Watch:

https://github.com/sharandac/My-TTGO-Watch/issues/374#issuecomment-1244544427

So maybe LillyGo T-Watch can get 1-2+ days with 380mA or 400mA battery and ULP deep sleep capability?

As it stands I can not put the watch on at 8am and have it usable in the evening just 12 hours later( 8pm ).

dlarue, 2022-09-24:
I charged to 98% this morning, changed min screen timeout from 15 to 5: `#define DISPLAY_MIN_TIMEOUT         5`
and only checked the battery level 4 times(minimal screen time) so most all battery usage has been while resting.
Yet, after 6 hours it is already at 66%.

A far better sleeping system needs to be implemented and using ULP might be a good solution.

lewisxhe, 2022-09-29:
Later if there is an update I will pay attention to this

closed 2023-02-06 by lewisxhe