Xinyuan-LilyGO/TTGO_TWatch_Library #174: Unable to upload any example of the library

Kaki-In, 2022-09-28:
Hi, 
I can upload a program to my watch, and this one works successfully : 
 
``` cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.printf("Test\n");
  delay(1000);
}
```

But, when I try to upload a TTGO_TWatch_Library example program, it is not uploaded to the card :
 - While the compilation prcess, I have few cpp errors.
 - At the end of the compilation, it blocks after saying which libraries are used for something as one minute and then says "Error : connection has been disposed".
 - Immediatly before and after this message, the card is still available.

I have set the #define LILYGO_WATCH_2020_V3 at the begginning of the line and just installed the latest versions of the ESP32 cards json file and of the LILYGO_WATCH library. 

Can someone help me?

lewisxhe, 2022-09-29:
Please provide a screenshot

Kaki-In, 2022-10-02:
[There](https://www.flopcreation.fr/VideoGithubTTGOWatchIssue.mp4) is the video capture of what I get.

Kaki-In, 2022-10-02:
I did a simple syntax check on an errored message and got the same result : it seems that it itsn't about the watch, but the library itself.

lewisxhe, 2022-10-03:
[image]

It's normal on windows, I think this may have something to do with the version of arduino-esp32, what version of core are you using?

Kaki-In, 2022-10-07:
I actually am on Ubuntu

Kaki-In, 2022-10-07:
cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.1 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

lewisxhe, 2022-10-10:
[image]

I tried in the newly installed UBuntu22.04.1 LTS and found no errors

lewisxhe, 2022-10-10:
This is a video recorder
https://youtu.be/BzbEkUnE7C4

Kaki-In, 2022-11-27:
Go into File -> Preferences
In the "Compiler Warning" Section, select "All"

Retry to compile

Installing a second version I had a lot of same errors...

Kaki-In, 2022-11-27:
[image]

Kaki-In, 2022-11-27:
( a second version of ubuntu, I am using a vm)

Kaki-In, 2022-11-27:
Ok I just disactived this and now it works!

Thanks for your help

Kaki-In, 2022-11-27:
Just a little question quickly... Where is the main Operating System of the watch?

lewisxhe, 2023-02-06:
If you have any questions, please restart

closed 2023-02-06 by lewisxhe