Xinyuan-LilyGO/TTGO_TWatch_Library #179: Latest factory firmware sharandac/My-TTGO-Watch uses VS Code is there any info on how to use this with Arduino IDE?

iweave1, 2022-11-18:
Latest factory firmware sharandac/My-TTGO-Watch uses VS Code is there any info on how to use this with Arduino IDE? 
VS Code is not something that I want to use. If I am forced to use it for this watch, it will take away my desire to work with it.

saubhaagyapurna, 2022-11-20:
I found the answer. Though I still have to install vs code and platformio to convert it.
It would be nice if this was something that you could do for us in the community that still use Arduino IDE.

https://runningdeveloper.com/blog/platformio-project-to-arduino-ide

lewisxhe, 2023-02-06:
My-TTGO-Watch is written by sharandac, and arduinoide cannot implement the functions of platformio

closed 2023-02-06 by lewisxhe