Xinyuan-LilyGO/TTGO_TWatch_Library #182: Any tips about how to use sprites ?

DarkMulet, 2022-12-10:
Hello there I have three questions today, 
First: how can i print sprites on the T Watch ?
I searched a bit and think I need to define the sprite first using something like 
'''
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);
'''
then in void setup 
'''
sprite.createSprite();
'''
Second question is How to rotate sprites of a given angle with the rotation point beeing the start point of the sprite I guess
Third question is what is the most efficient way to encode external images as sprites ?

I searched a bit about sprites and the watch on google but it seem it's not the same use as classig TTGO Displays so I need a bit help there
thank you in advance a keep working ^^

lewisxhe, 2022-12-12:
see there https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/TFT_eSPI/Sprite

codefaux, 2023-01-30:
As a more general note, direct from the `TFT_eSPI.h` library header, two important notes for crafting sprites externally;

```
// Next is a special 16 bit colour value that encodes to 8 bits
// and will then decode back to the same 16 bit value.
// Convenient for 8 bit and 16 bit transparent sprites.
#define TFT_TRANSPARENT 0x0120 // This is actually a dark green

// Default palette for 4 bit colour sprites
static const uint16_t default_4bit_palette[] PROGMEM = {
    TFT_BLACK,    //  0  ^
    TFT_BROWN,    //  1  |
    TFT_RED,      //  2  |
    TFT_ORANGE,   //  3  |
    TFT_YELLOW,   //  4  Colours 0-9 follow the resistor colour code!
    TFT_GREEN,    //  5  |
    TFT_BLUE,     //  6  |
    TFT_PURPLE,   //  7  |
    TFT_DARKGREY, //  8  |
    TFT_WHITE,    //  9  v
    TFT_CYAN,     // 10  Blue+green mix
    TFT_MAGENTA,  // 11  Blue+red mix
    TFT_MAROON,   // 12  Darker red colour
    TFT_DARKGREEN,// 13  Darker green colour
    TFT_NAVY,     // 14  Darker blue colour
    TFT_PINK      // 15
};
```

So, for both 16-bit and 8-bit color, use TFT_TRANSPARENT for a color intended to be transparent. This is 0x0120 when encoding in 565, and 0x002400 when encoding in RGB 24-bit color.
If you master a sprite on the PC and use 0x002400 as your transparency key, after encoding the same sprite into 565 color, any sprite functions which support transparency keying will respect it. You can specify an alternate transparency color in these cases, but this is a handy inbuilt default.

Also detailed is the "default" color palette, which can be used when externally mastering sprites dedicated for this display, to avoid shifts in color tone due to bit-depth misalignment and a more shallow available spectrum of colors.

lewisxhe, 2023-02-06:
If there is a problem, please reopen it

closed 2023-02-06 by lewisxhe