Xinyuan-LilyGO/TTGO_TWatch_Library #183: Any tips about how to use images ?

DarkMulet, 2022-12-12:
Hello there it's me again, I started to understand how the watch work because studying it non stop for the last few days
I want now to show images on the watch but I have two problems here:
I only found examples using lvgl to show images.

First : is there a way to load images in sprites instead of lvgl ? It would draw must faster and easier to use with sprites I think.
Second : The only working converter from images to c I've found is the official lvgl converter but it only convert to Hex2 so no colors are availables in the images, how can I show colorfull images on the watch ?

The purpose here is to start making Applications Icons like Apple Watch, obviously using colors.

Thank you in advance.

lewisxhe, 2022-12-13:
see there https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/TFT_eSPI/Sprite/Sprite_image_8bit

ImageConverter  : http://rinkydinkelectronics.com/t_imageconverter565.php

lewisxhe, 2023-02-06:
If there is a problem, please reopen it

closed 2023-02-06 by lewisxhe