Xinyuan-LilyGO/TTGO_TWatch_Library #184: L76K Gps Module Garbled message

Skillie-ZA, 2022-12-22:
Hi,
I have a TTGO watch V2 with L76K GPS module.
Since i got it the GPS has never worked
Running the GPSDisplay.INO this is my output:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
L76K GNSS init succeeded, using L76K GNSS Module
L76K GNSS init succeeded, using L76K GNSS Module
L76K GNSS init succeeded, using L76K GNSS Module
L76K GNSS init succeeded, using L76K GNSS Module
L76K GNSS init succeeded, using L76K GNSS Module

$GNRMC,,V,,,,,,,,,,N,V*37
DATE Fix Age=15ms Raw=720997707 Year=2007 Month=77 Day=163
TIME Fix Age=23ms Raw=423042734 Hour=167 Minute=4 Second=27 Hundredths=34
$GNGGA,,,,,,0,00,25.5,,,,,,*64
$GNRMC,,V,,,,,,,,,,N,V*37
DATE Fix Age=7ms Raw=720997707 Year=2007 Month=77 Day=163
TIME Fix Age=15ms Raw=423042734 Hour=167 Minute=4 Second=27 Hundredths=34
SATELLITES Fix Age=19ms Value=0
HDOP       Fix Age=24ms Value=2550

My Thinking is that the GPS is in SIRF mode and needs to be set to NMEA, But I cant seem to find a Command to do that for Arduino? 

Many Thanks for your Great Work!!!

lewisxhe, 2022-12-23:
I didn't find the SIRF in the user manual of the L76K. It should not be supported. According to the output log, it can be seen that it is working normally. I need to doubt whether the GPS antenna is normal. I have  a user manual in Chinese, but not in English

[Quectel_L76KL26K_GNSS_协议规范_V1.2.pdf](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/files/10293606/Quectel_L76KL26K_GNSS_._V1.2.pdf)

Skillie-ZA, 2022-12-23:
Thanks, I suppose it's the Antenna, As it hasnt worked since day1 even with all the Stock and NON stock Firmware's and Applications.
It's ok to close this issue !
Thanks for  your speedy response!

lewisxhe, 2023-02-06:
If there is a problem, please reopen it

closed 2023-02-06 by lewisxhe