Xinyuan-LilyGO/TTGO_TWatch_Library #187: write $PCAS commands

NDC-SC, 2023-02-05:
In the Arduino IDE Examples> TTGO TWatch> BasicUnit> TwatcV2Special> GPSDisplay sketch, there are several write instructions to $PCAS commands.
write("$PCAS06,0*1B\r\n");  // version information
write("$PCAS04,5*1C\r\n"); // satellite reception GPS and GLONASS
write("$PCAS03,1,0,0,0,1,0,0,.... // define which NMEA messages to receive
write("$PCAS11,3*1E\r\n"); // mode ?

Can you please provide a link to information on the command parameters, including construction of the checksum.
For example, resetting PCAS03 to include all NMEA messages.
Is the instruction write("$PCAS03,1,1,1,1,1,1,1,1,0,1,,,1,0*02\r\n"); correct ?

The Quectel_L76KL26K_GNSS (in Chinese) document does not include PCAS06 and PCAS11
The Quectel_L76-LBL26-LBLC86L_GNSS_Protocol_Specification_V1.1 does not include any PCAS commands.

Thanks

lewisxhe, 2023-02-06:
This is the manual
[CASIC_ProtocolSpecification.pdf](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/files/10636690/CASIC_ProtocolSpecification.pdf)

NDC-SC, 2023-02-06:
Thanks for the link.
Do you have details of PCAS11 as used in the sketch: write("$PCAS11,3*1E\r\n");

lewisxhe, 2023-02-07:
I can't find the command containing PCAS11. This may be an error

NDC-SC, 2023-02-07:
See line 55 of the GPSDisplay sketch in Arduino IDE Examples> TTGO TWatch> BasicUnit> TwatcV2Special

closed 2023-02-11 by lewisxhe

NDC-SC, 2023-02-12:
Thank you for providing details of writing $PCAS commands.