Xinyuan-LilyGO/TTGO_TWatch_Library #188: Bluetooth and GPS doesn't work on the T-WATCH-2020 V2

itsalisa, 2023-02-22:
Hello,

I have recently bought a T-Watch-2020 V2 and I'm trying to connect it to my smartphone via Bluetooth to test connectivity for my project. It seems like both Bluetooth and GPS doesn't work but however the watch can connect to the WiFi.

I can't figure out what the issue could be. Does anyone have any suggestions?

CustomCheat, 2023-02-22:
I have similar issues - My GPS doesn't want connect to any towers. I have other issues with my watch, but it seems I need to reflash the watch, u can try that.

itsalisa, 2023-02-22:
Yeah, I have reflahed it but still no connection. My guess is the problem is within the firmware itself.

CustomCheat, 2023-02-22:
The only other solution I found is that it may be the hardware, as mentioned here: https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/issues/141 But I really don't know, so I hope someone still does manage the issues here.

lewisxhe, 2023-02-23:
I suspect that the circuit of the hardware antenna is short-circuited. If you have a multimeter, please measure whether the antenna and GND are connected together

ThatOneDelta, 2023-03-24:
had the same issue with GPS. In my case it was pretty much what [lewisxhe](https://github.com/lewisxhe) descibed

as they have pinted out, you can check if the soldering points of the antenna are shorted to ground. If they are, you can try to open up the case, unscrew the 4 screws (be careful with the motor cables, they are very fragile) and carefully remove the band (the one with the big oval hole and the metal rivet)

you will find the antenna PCB. The problem is, that the solering points touch the aluminum case of the watch. To fix this, you can cover them with a piece of tape (ideally kapton tape) and reassemble. Do take your time with this, since everything is super fragile when dieassembled

after covering the soldering points with tape, GPS worked just fine for me