Xinyuan-LilyGO/TTGO_TWatch_Library #189: After uploading, watch bootloops for ~20 minutes

CustomCheat, 2023-02-23:
After i upload my sketch, the watch sometimes does restart and works just fine, but after like 3-4 uploads it is in a bootloop for ~20 minutes or so, after that it boots up. Output from serial monitor: First it does this: 
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xffff03fc,len:-65281
1150 mmu set 00010000, pos 00010000
`
After that it just spams this:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
`
I did reflash the watch, but it didn't help.

lewisxhe, 2023-02-23:
Will it restart automatically after 10~20 minutes of normal operation?
What firmware do you use? Can you run the simple firmware and try whether it will restart automatically.
For your log above, if it appears after normal operation, it is likely that the internal data of flash is lost or damaged.
If you can, please try to brush in a simple test program to test whether it can run normally for a period of time. If not, it can be determined that there is a problem with the watch

CustomCheat, 2023-02-23:
I have the V2 watch and i flashed the V2 software, using a windows VM. One thing that stood out was that the FINISHED was not in green, but in blue. Does that effect anything? I did erase the watch before flashing. 
I tried upload the GPS display sketch and it does upload and run normally (mind you it shows invalid location, but that doesnt matter for now) But after the second upload of the same sketch, the watch doesnt show anything and spams the RTCWDT_RTC_RESET and some mmu set sometimes. Sometimes rebooting solves it, sometimes it doesnt work for ~10 minutes, then it works like nothing happened. It does this when uploading a sketch, after that everything works. I had the display of the watch turned on for 25 minutes and it works, so the only problem is the uploading.

CustomCheat, 2023-03-10:
Seller sent a new watch, will see if the issues are still there.

CustomCheat, 2023-03-13:
Watch seems to work. Closing, for now.

closed 2023-03-13 by CustomCheat