Xinyuan-LilyGO/TTGO_TWatch_Library #190: LilyPi USB

noranger, 2023-02-26:
How can the USB ports switched on constantly.  digitalWrite(EXTERN_USB_EN, HIGH); or ttgo->turnOffUSB(); only works for a second after booting.

lewisxhe, 2023-02-27:
TurnOnUSB only sets GPIO33 to HIGH. If you operate on IO33 externally, it may be affected

noranger, 2023-02-27:
Gpio 33 is reserved for UART_TX, but i need power. Gpio33 is not used by turnOnUSB().

lewisxhe, 2023-02-27:
I made a mistake. USB enables GPIO4, while LilyPi's relay uses GPIO32
Do you use GPIO4 anywhere else?

noranger, 2023-02-27:
No

lewisxhe, 2023-02-27:
Please use the simplest test code to just set GPIO4 to HIGH and see if USB will maintain output
```
void setup()
{
pinMode(4,OUTPUT);
digitalWrite(4,HIGH);
}

void loop(){
}
```

noranger, 2023-02-27:
Only Power after booting for 20ms.
My Code for Setup.

//boardmanager esp32 Wrovwer Modul (LilyPi)

#define LILYGO_LILYPI_V1
#define LILYGO_BLOCK_ST7796S_MODULE         //Use ST7796S
#define LILYGO_WATCH_LVGL                   //To use LVGL, you need to enable the macro LVGL
#include <LilyGoWatch.h>

#include <Serial_CAN_Module.h>
#include <SoftwareSerial.h>

Serial_CAN can;

#define can_tx  33           // tx of serial can module
#define can_rx  12          // rx of serial can module
#define CAN_RATE_250    15

unsigned char flagRecv = 0;
unsigned long id = 0;
unsigned char rxBuf[8];
char str[20];

int temp=0; int temp2=0; int kreis1=0;int kreis2=0;int kreis3=0;int gang=125;int retarder=0;int mbremse=0;int gewicht=0;int partikelf=0;
char Motortemp[4]; char Intarder[4]; char Gear[4]; char Exhaustbrake[4];

TTGOClass *ttgo;

void setup()
{
  Serial.begin(115200);SPI.begin();

  //pinMode(EXTERN_USB_EN, OUTPUT);
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);Serial.println("USB On");

    //Get watch instance
    ttgo = TTGOClass::getWatch();
    ttgo->begin(); 
    //ttgo->openBL();// Turn on the backlight

    ttgo->tft->setRotation(1); ttgo->tft->setTextFont(4);
    ttgo->tft->fillScreen(TFT_BLACK);ttgo->tft->setTextColor(TFT_WHITE,TFT_BLACK);

    can.begin(can_tx, can_rx, 115200);      // tx, rx

    if(can.canRate(CAN_RATE_250))
    {Serial.println("set can rate ok");}
    else
    {Serial.println("set can rate fail");}

.......

lewisxhe, 2023-02-27:
There is no problem with the code. Are you connected to an external load? Please remove the USB load and try to see if this happens. I suspect that the current of the external load is too large, which causes the USB current-limiting start

noranger, 2023-02-27:
Yes. Now the Usb power is on all ports. I testet it with a led. 3v and 5v switching works too.
 The Serialcanbusboard is now making problems. Thanks for your fast help.

closed 2023-02-27 by noranger