Xinyuan-LilyGO/TTGO_TWatch_Library #195: Can I program the T-watch 2021 with Python only?

Angelo904, 2023-06-18:
I have recently purchased a T-watch 2021 and, since Circuit Python is the only language I am familiar with, I would like to find out if I can program it just using python, and what "kind" of python I would beed to use (can I use Circuitpython or it has to be Micropython?).

I looked around on this GitHub page and it seems that all the software available for the T-watch 2021 is in Arduino / C++

However I have read in several different places (including the Lilygo open smartwatch product page) that it can be programmed with pyhton.

Can anybody clarify whether it can actually be done and what I would need to do? Where can I find the python libraries/modules? Can I just write any python code on Platform IO and upload it to the board?

Thanks in advance!
Angelo

VynDragon, 2023-07-02:
Yes, you just need to get or make drivers.

For example, you can use LewisHe's libraries that were sponsored by lilygo: 
- https://github.com/lewisxhe/XPowersLib/tree/master/Micropython/src (this one for the AXP serie, compatible with both circuitpython and micropython, though im not sure the hardware it is for is on the t-watch 2021)
- https://github.com/lewisxhe/PCF8563_PythonLibrary for RTC 

Any other library or driver you need could be found using <peripheral name> + circuitpython.

Circuitpython will work as well as micropython, but you might need to adapt the code if it is not made for circuitpython

I have written a firmware for the *2020* version of the watch, in *micropython*, but i invite you to look at it (in Hardware.py) as there are a lot of subtelties that come with some of the hardware, like the BMA423 (that is also present on T-WATCH 2021):  https://github.com/VynDragon/WPOS2/blob/master/system/Hardware.py

Angelo904, 2023-07-06:
> Yes, you just need to get or make drivers.
>
> For example, you can use LewisHe's libraries that were sponsored by lilygo:
>
> * https://github.com/lewisxhe/XPowersLib/tree/master/Micropython/src (this [...]

@VynDragon, many thanks for your very useful answer. In the meantime, I have started working on some Arduino code. I guess I’ll keep going this way but it is reassuring to know that if it gets too difficult for me I have the option to switch to python. I’ll keep you posted. Angelo