Xinyuan-LilyGO/TTGO_TWatch_Library #197: S3 watch completly missing axp2101 power domains mappings documentation

VynDragon, 2023-07-02:
The schematics are incomplete/wrong, and with both them and the arduino code I have not been able to determine the roles of each LDO and other power domains.

Please document what powers what like for the original watches.

VynDragon, 2023-07-02:
Here are my notes about them though:

| Power Domain  | Role |
| --| --|
| ALDO1  | MS412FE RTC backup battery (3.1-3.3v)  |
| ALDO2 | Backlight  |
|ALDO3|3V3 for FT6336 and st7889|
|ALDO4|sx1262|
|BLDO2|DRV2605 Enable pin|
|DC5|FT6336 RST pin|
|VRTC| Pullup for AXP2101 IRQs (not working, setup pin 21 as pull up)|

Please correct what i missed

lewisxhe, 2023-07-03:
Modifications have been made to your answer

closed 2023-07-03 by VynDragon