Xinyuan-LilyGO/TTGO_TWatch_Library #204: Battery - where is a replacement 400mah battery?

dlarue, 2023-08-25:
I have two 2020 v1 watches and I can't get more than 8 hours out of either but their batteries are old and they are 380mah.
Where can I get the later model's 400mAh battery?  Even a model number might be helpful. The 380mah model numbers are HY-QA

VynDragon, 2023-08-31:
It wont fit unless you remove the connector.
8 Hours is also very low even with a well toasted battery, you might want to optimize your software.
[image]
[image]

VynDragon, 2023-08-31:
Also the hole in the case is 31 mm x 24mm

dlarue, 2023-08-31:
Is there somewhere where I can order the stock 380mA battery then?

Also, the software is the default recommended software, My-TTGO-Watch and I turned off Wifi, Bluetooth and vibration so I thought it was running at pretty low CPU. I even lowered the CPU frequency. This is why I thought it was due to old batteries.

VynDragon, 2023-08-31:
Yes, I ordered some 'LQ-S1' batteries from ali and they fit exactly the same as 'HY-QA' batteries: https://fr.aliexpress.com/item/1005003566854829.html

VynDragon, 2023-08-31:
I managed to remove a sticker on my S3, **502530** is the exact cell size.

dlarue, 2023-08-31:
> Yes, I ordered some 'LQ-S1' batteries from ali and they fit exactly the same as 'HY-QA' batteries: https://fr.aliexpress.com/item/1005003566854829.html [image] (and notice how the HY-QA didnt even properly fit in that V2 in the first place lol)

unfortunately no longer available. I will search for batteries for the watch version specified on that page.

dlarue, 2023-08-31:
I've read that leaving it charging over night is a bad idea and the battery will over charge and have weaker capacity.
Have you experienced that or know if the onboard charger does not over charge if left connected to USB power?

VynDragon, 2023-08-31:
It depends on how you set it up, you are in control, look up the datasheet for the power control chip (axp202 on V1,V2,V3 axp2101 on S3)
By default it appears to cut charging out when the battery reaches the set charge voltage, and resume charging when it drops beyond a certain voltage.

dlarue, 2023-09-04:
@VynDragon  I was able to get the new LQ-S1 380mAh battery into the T-Watch 2020 V1 but had to remove the label as it was just too tight otherwise.  With this new battery I have already seen it work for more than 12 hours on its first charge and that is 50% longer than the original batteries which only lasted for less than 8 hours. I have 2 T-Watch 2020 v1 watches, one new, one old and even the new one only got less than 8 hours of battery life with lots of steps to lower battery use(low CPU clock speed, short LCD ON time, Vibrate=off, Bluetooth=off, Wifi=off, etc. This is very good news!

Regarding the software I'm using, it is the recommended/Stock software(https://github.com/sharandac/My-TTGO-Watch ) and I will be sure to only charge with the watch powered ON since you say it is under software control.

VynDragon, 2023-09-04:
This is not how the software settings work, they are set then kept.
Also this is bad autonomy, my firmware reaches 48h easily on V1...

dlarue, 2023-09-04:
Thank you. I will look at the battery charger chip datasheet and see how it is configured to see if there is any connection/configuration by the esp32 firmware so that it can charge properly when turned off. 

48h?  Is this running the default watch application(My-TTGO-Watch) or some other application? If some other can you point my to it and I will try that.  

Also, because the watch battery indicator is not reliable, I don't have a way to know when the battery is low. I believe I ran it to 65% when it was at 12h on my first test.

VynDragon, 2023-09-19:
Reporting one of my LQ-S1 just went pillow, despite having protection.
I also received 102530 batteries and they do fit as expected, ill put up a pull request when im done with the shell file.

dlarue, 2023-09-19:
That's generally how the Lithium batteries reach EOL. My NVidia tablet just did that after about 5 years.
As you mentioned, a "good" battery in a T-Watch lasts for a couple of days. I've put away the old Casio and started wearing the T-Watch daily and it's been holding up well as long as I keep Wifi off.  My plan now is to work on a test app which will wake up WiFi, send a message and put WiFi back to sleep and see how well that works as opposed to trying to keep a WiFi connection going all the time.

AlexeiScherbakov, 2024-07-23:
Found this issue, 502530 is battery size on T-Watch S3