Xinyuan-LilyGO/TTGO_TWatch_Library #207: esp_vad.h: No such file or directory

robgerpin, 2023-09-07:
Error compiling Factory demo example code:

Compilation error: esp_vad.h: No such file or directory

#include <esp_vad.h>

What is the library dependency?

alviso, 2023-09-09:
Same issue here. Please comment

lewisxhe, 2023-09-09:
Please update esp32_core to the latest, this is the built-in function of esp

alviso, 2023-09-09:
Thanks. Funny enough, switching to 2.0.5 solved it for me (previously 2.0.12)

lewisxhe, 2023-09-09:
esp-sr exists on the master branch, but I don't know why it was removed in the release version. I will mark the version

https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32s3/include

lewisxhe, 2023-09-09:
I don't understand why esp-sr has been removed after 2.0.9. Please use version 2.0.9 and I will investigate later.

ddlsmurf, 2023-09-23:
So I didn't test 2.0.9, but I had this error with .12, so rolled back. For my fellow googlers, here's the snippet to add to your platformio.ini file to do so fo 2.0.6 which does fix this issue (too - if someone can try 2.0.9 and report that would be much appreciated):

```ini
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git@2.0.6
```

jeff-burright, 2023-10-05:
> So I didn't test 2.0.9, but I had this error with .12, so rolled back. For my fellow googlers, here's the snippet to add to your platformio.ini file to do so fo 2.0.6 which does fix this issue (too [...]

I am having the same esp_vad.h issue, but when I tried this quoted fix it threw me an exception like so:

PackageException: Package version 3.0.0+sha.6f7a1ca doesn't satisfy requirements 2.0.9 based on PackageMetadata <type=tool name=framework-arduinoespressif32 version=3.0.0+sha.6f7a1ca spec={'owner': 'platformio', 'id': None, 'name': 'framework-arduinoespressif32', 'requirements': '2.0.9', 'uri': 'git+https://github.com/espressif/arduino-esp32.git'}

jhohannfabian, 2023-10-28:
Hi, I have the same problem. If you reduced it to the proposed version, it compiles, but does not record the voice. In the Arduino IDE 2.014 version, they do not include the esp_vad.h library. I tried adding it, pasting it, and couldn't get it to work.