Xinyuan-LilyGO/TTGO_TWatch_Library #209: T-Watch-S3 xtensa-esp32s3-elf-g++: not found

ukoda, 2023-09-14:
I am trying to build the the T-Watch-S3 example on a Linux Mint 21.1 x86_64 system but it does not appear to be finding the compiler.

Before attempting to work with the T-Watch-S3 I set up VC Code with the PlatformIO extension and was able to successfully build and flash an older T-Watch 2020.  However the build fails for the T-Watch-S3 which uses a different compiler.

I followed the (https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3#1%EF%B8%8F%E2%83%A3-platformio-quick-start-) steps but suspect there is either an additional undocumented step I need to follow or the project is not set up to ensure the needed tool chain is installed.  Specifically I:
1. Checked I could build and flash the older T-Watch 2020 with VS Code/PaltformIO with the (https://github.com/sharandac/My-TTGO-Watch) project.
2. Exited VS Code and cloned this repo.
3. Checked out the `t-watch-s3` branch.
4. Started VS Code and Open Folder -> select the TTGO_TWatch_Library directory of the cloned repo.
5. The build failed with an `sh: 1: xtensa-esp32s3-elf-g++: not found` error.
6. Switched the Project Environment to `env:twatch-3`.
7. The build failed with an `sh: 1: xtensa-esp32s3-elf-g++: not found` error.

The full output seen is:

```

Resolving twatch-s3 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing twatch-s3 (platform: espressif32; framework: arduino; board: LilyGoWatch-S3)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/LilyGoWatch-S3.html
PLATFORM: Espressif 32 (3.3.0) > LilyGo T-Watch S3 (16M Flash 8M OPI PSRAM )
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting RTC_TimeLib.ino
sh: 1: xtensa-esp32s3-elf-g++: not found
*** Error 127
AssertionError: :
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/david/.platformio/platforms/espressif32/builder/main.py", line 228:
    target_elf = env.BuildProgram()
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 339:
    env.ConvertInoToCpp()
  File "/home/david/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/pioino.py", line 237:
    out_file = c.convert(ino_nodes)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/pioino.py", line 85:
    return self.process(contents)
  File "/home/david/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/pioino.py", line 106:
    assert self._gcc_preprocess(contents, out_file)
========================== [FAILED] Took 0.87 seconds ==========================

```
I'm new to PlatformIO so I am not sure where or how it normally installs and finds tool chains that a project needs.

Please let me know if additional information is needed

brucekissinger, 2023-09-14:
For what it's worth, I had a similar problem with the same error message trying to use PlatformIO on Windows 10.  I carefully followed the instructions several different times and tried to resolve the issue over several days but eventually gave up.  It was very frustrating.

I tried installing the Espressif IDE and noticed that the compiler was installed as part of that software.  On my Windows 10 machine, it was located in C:\Users\bruce\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin

ukoda, 2023-09-14:
@brucekissinger  Yes, I have the needed tool chain on my system from a prior Espressif environment, but it is not on path.  Putting it on path does fix the problem, for me, but I think the root cause should be determined and addressed so the next newbie doesn't have a negative experience too.

ukoda, 2023-09-14:
Having kludged the project to attempt build by putting a suitable tool chain on path it is now throwing a build error about `pins_arduino.h` being missing.  While I could create a new issue for that it does feel like it related to this issue and I do continue to wonder if the set up of this project is making undocumented assumptions about what is already present on a system and therefore is failing to install all the needed dependencies.

ukoda, 2023-09-14:
Progress.  While the `pins_arduino.h` issue remains I have resolved the original compiler issue.  From reading (https://community.platformio.org/t/fatal-error-pins-arduino-h-no-such-file-or-directory/19634/13) I figured I may have stale PlatformIO files for years ago when I was abandon PlatformIO at the time it moved from open source to a closed model.  In my case I closed VS Code, removed the tool chain I had manually added to the path and removed:

- ~/.platformio/packages/framework-arduinoespressif32*
- ~/.platformio/packages/toolchain-xtensa32/
- ~/.platformio/packages/tool-esptoolpy/

After doing that I was able to build to the point it couldn't find the pins_arduino.h file.

closed 2023-09-14 by ukoda