Xinyuan-LilyGO/TTGO_TWatch_Library #212: LoraWAN on T-Watch S3

h0ryz0n, 2023-10-19:
we have a T-Watch S3 but we are having issues to understand how to make LoRaWAN work with our LNS (chirpstack), because there are LoRa but no LoRaWAN examples in the offical repository: https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3/examples/radio
could you explain how to make LoRaWAN work with T-Watch S3 and give a working example?

lewisxhe, 2023-10-23:
There is no LoRaWAN library available for SX1262. I can't help you.

h0ryz0n, 2023-10-25:
there is actually a library available for SX1262 on [RadioLib](https://github.com/jgromes/RadioLib) repository, which you used for examples, but I couldn't figure out how to make it work for the S3

lewisxhe, 2023-10-25:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3/examples/radio/LoRaWAN

Thank you for your notice. When writing the example, the RadioLib version was 6.0.0. I have added the corresponding example, but I cannot test it now. If you can, please tell me the test results.
Please remember that RadioLib needs to be upgraded to 6.2.0 before it can be used