Xinyuan-LilyGO/TTGO_TWatch_Library #214: T-WATCH T-2020 S3 only shows the *modem* tty device, not the serial one.

antirez, 2023-10-26:
Hi, I'm using MacOS 13.5.1. I can see the modem device in `/dev/tty.usbmodem1234561` but this is not good for programming, as it translates the escapes of `esptool`. I installed the CH34X driver you recommend, it can see all my other Lilygo devices (T3, T-BEAM) as serial interfaces, but this one only shows the modem that can't be used correctly. What should I do?

lewisxhe, 2023-10-28:
t-watch-s3 uses the built-in USB of esp32s3 to upload sketches, so there is no need to install the ch34x driver. Have you tried the method mentioned in the [FAQ](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3#6%EF%B8%8F%E2%83%A3-faq-)?

antirez, 2024-01-27:
Thanks @lewisxhe, the problem is that I was trying to install MicroPython. And in the end I was able to, however after I installed MicroPython, now the device is no longer accessible with 'esptools'. This is what I did:

```
esptool.py --chip esp32s3 --port /dev/tty.usbmodem* erase_flash
esptool.py --chip esp32s3 --port /dev/tty.usbmodem* write_flash -z 0 ESP32_GENERIC_S3-20231005-v1.21.0.bin
```

Now I can reach MicroPython, but now if I want to erase the flash again I can't, even if MicroPython works:

```
sudo esptool --chip esp32s3 --port /dev/ttyACM0 erase_flash
esptool.py v4.6.2
Serial port /dev/ttyACM0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Do I need to press the "boot" switch that there is inside the device? Thanks.
```

lewisxhe, 2024-01-27:
If you write MicroPython it generates a virtual port instead of a real physical download port for esp32 and if you want to re-wipe you can only put the watch into download mode

antirez, 2024-01-27:
Thank you, understood. Closing this issue and opening a new one about a different issue. Thanks.

closed 2024-01-27 by antirez