Xinyuan-LilyGO/TTGO_TWatch_Library #223: Compiling LiliGoLib.h missing file + error uploading empty sketch

chritron, 2024-01-26:
Hi,

I have encountered two things when trying to program a new TWatch S3, using Arduino IDE 2.2.1 with ESP32 2.0.14, board ESP32S3 Dev Module.

1. When trying to compile one of the example sketches, I get the following message:
..Documents\Arduino\libraries\TTGO_TWatch_Library-t-watch-s3\src/LilyGoLib.h:25:10: fatal error: TouchDrvFT6X36.hpp: No such file or directory
 #include <TouchDrvFT6X36.hpp>

I cannot find this file anywhere.

2. When trying to upload just an empty sketch, I get the following error:

 A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

Have tried with different COM ports and cables and verified that it works on another (Adafruit) ESP32 board).

lewisxhe, 2024-01-26:
Hi,
1. Have you installed [T-Watch-Deps](https://github.com/Xinyuan-LilyGO/T-Watch-Deps) according to the method in the README?
A copy of the library LilyGoLib depends on can be found here [T-Watch-Deps](https://github.com/Xinyuan-LilyGO/T-Watch-Deps)
2. When the library was written, the esp32 core was 2.0.9 and 2.0.14. Some examples may have compilation errors, and there is currently no time to adapt.
If there is an error in uploading the sketch, you can put the watch into download mode according to the README [FAQ](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/t-watch-s3/README.MD#6%EF%B8%8F%E2%83%A3-faq-) section.

chritron, 2024-01-26:
Thanks a lot for the quick response. 

After now having installed the T-Watch-Deps according to the method in the README, I am only getting the following error when compiling:

..Documents\Arduino\libraries\lvgl\src\osal\lv_windows.c:14:10: fatal error: process.h: No such file or directory
 #include <process.h>
          ^~~~~~~~~~~
compilation terminated.

exit status 1

When trying to upload, I have tried to follow the FAQ point 1 about that, and the device is identified with a COM port in the Device Manager, as a USB Serial Device with the same port number as configured in the IDE. The board is configured in the IDE (under Tools) as listed in the README file. I am still getting the following error when trying to upload an empty sketch:

Sketch uses 235761 bytes (7%) of program storage space. Maximum is 3145728 bytes.
Global variables use 18328 bytes (5%) of dynamic memory, leaving 309352 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM72
Connecting...

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

Thank you for any advice

lewisxhe, 2024-01-27:
I recorded a video on how to get started, you can refer to it.
https://youtu.be/3yxTeY2orLA

lewisxhe, 2024-01-27:
If you encounter an error when uploading, it may be related to the version of esptool. Versions less than 4.6.0 may encounter errors. If it is less than 4.6.0, you can try to put the watch into download mode first. I already mentioned above how to put the watch into download mode

lewisxhe, 2024-01-27:
I recorded a demonstration video on how to upgrade esptool. I hope it will be helpful to you.
https://youtu.be/TZEnynJfAjY

chritron, 2024-01-27:
Thank you so much for making these videos.

I have accordingly upgraded the esptool to v4.7.0, but I still receive the
same error message:

[image: image.png]
I did not understand the instruction of how to put the watch into download
mode, according to the FAQ:

   1. Make sure that the T-Watch is turned on, you can check it according
   to the following method, open the computer device manager, check the port,
   plug the USB port into the computer, and if the new COM device is
   displayed, it has been turned on, if it is not displayed, press the crown
   Press the button on the button for one second, and then the device port
   will pop up, click upload at this time

The COM device is correctly displayed in the Device Manager. I have tried
different ways to push the button for different durations before uploading,
but it did not resolve the error. Please let me know if there is anything I
did not understand.

With respect to compilation of example sketches, i noted the following from
the first video:

1. Use ESP32 v2.0.9
2. Unpack the TTGO_TWatch_Library-t-watch-s3 library and copy the folder to
the Arduino/Libraries folder (This was already installed, but did it again)
3. Unpack the contents of the T-Watch-Deps-master into the
Arduino/Libraries folder (This was already installed, but did it again)
4. Now having a similar directory structure as in your video in 1:55
5. Open Arduino IDE (2.2.1) and opening the same example as you (all free
fonts)
6. Selecting ESP32S3 Dev Module and the COM port that is recognized in
Device Manager (USB Serial Device)
7. Verifying that ts is the 2.0.9 version of ESP32 installed in board
manager.

Now trying to compile. The compilation gets to about 15% and the progress
bar is at this stage for several minutes without any output (see picture).
[image: image.png]
Eventually the compilation finishes with the following message:
[image: image.png]

So these errors seem to persist now. Do you know of anything else I can
try?

Thank you very much for all support.

lewisxhe, 2024-01-27:
If you are in doubt about how to enter download mode, then check the following steps
1. Open the back cover of the watch and take out the battery because the buttons are covered by the battery.

1. Connect the USB to the watch. Can you see the serial port in the computer manager?

If the port is visible, then keep the USB connected and press the crown button (above USB) until the Computer Manager serial port disappears

If the port is not visible, press the crown button for two seconds and see if the serial port is visible

2. Press the button pointed by the arrow in the picture below and keep pressing it

[image]

3. Press the button on the crown again until the serial port is visible in Computer Manager
4. Select the port in ArduinoIDE and click Upload Sketch

lewisxhe, 2024-01-27:
> Thank you so much for making these videos. I have accordingly upgraded the esptool to v4.7.0, but I still receive the same error message: [image: image.png] I did not understand the instruction of [...]

2~4 is correct as long as it has the same directory structure as in my video
If the compilation progress is stuck, this is a BUG of Arduino IDE
I can't view the picture you provided.
Currently, platformio is the most time-saving IDE. The compilation speed of Arduino IDE is too slow. If possible, try to use platformio.

chritron, 2024-01-27:
> esptool to v4.7.0, but I still receive the same error message: [image:
> image.png] I did not understand the instruction of how to put the watch

Thank you.I will try again tomorrow with this advice.

chritron, 2024-01-28:
Hi again,
I have now opened the back cover, remove the battery (keeping the
connection). The COM port is visible in the Device Manager (as a "USB
Serial Device"). Then I have tried to press down the button on the PCB as
shown in the picture, but nothing happens seemingly regardless of how long
I press the button (did for at least a minute while watching the Device
Manager).

About the directory structure, I have the identical directory structure as
in the video (at 1:55) in Arduino/Libraries, only except that I have
additional libraries there also.

The error message I get when compiling an example sketch is:
..

*Arduino\libraries\lvgl\src\osal\lv_windows.c:14:10: fatal error:
process.h: No such file or directory #include <process.h>
^~~~~~~~~~~compilation terminated.exit status 1*
*Compilation error: exit status 1*

The error message I get when trying to upload an empty sketch is:

* Sketch uses 228881 bytes (17%) of program storage space. Maximum is
1310720 bytes.*

*Global variables use 19152 bytes (5%) of dynamic memory, leaving 308528
bytes for local variables. Maximum is 327680 bytes.esptool.py v4.7.0Serial
port COM72Connecting...A serial exception error occurred: Write
timeoutNote: This error originates from pySerial. It is likely not a
problem with esptool, but with the hardware connection or drivers.For
troubleshooting steps visit:
https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
<https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html>Failed
uploading: uploading error: exit status 1*

Thank you for any advice

chritron, 2024-01-28:
Hi again,
I just have to add something.

I installed everything on another PC (Win11), and there I can upload an
empty sketch successfully to the Twatch. I notice that when connecting the
watch to this PC, it is recognized (in Arduino Board Info) as a ESP32S3 Dev
Module, while in the previous PC it is recognized as ESP32S3 CAM LCD.

I could also successfully compile an example sketch now, although it takes
several minutes to compile.

So now it seems to work, but I don't understand why it didnt work on the
main computer (with Win10).

Thank you for all the help.

lewisxhe, 2024-01-29:
```
*Arduino\libraries\lvgl\src\osal\lv_windows.c:14:10: fatal error:
process.h: No such file or directory #include <process.h>
^~~~~~~~~~~compilation terminated.exit status 1*
*Compilation error: exit status 1*

The error message I get when trying to upload an empty sketch is:
```
There is no lv_windows.c file in the lvgl directory. From the log output, I can't see where this file is referenced. Anyway, just solve it.