Xinyuan-LilyGO/TTGO_TWatch_Library #224: T-WATCH S3 - AXP2101 Init sequence

antirez, 2024-01-27:
Hello,

I'm trying to write some minimal implementation of libraries for MicroPython in order to support the T-WATCH S3.
I have installed MicroPython and soon I'll port my SX1276 driver to SX1262, however in order to do this I need to be able to power the whole device, and I need help about it.

I understand that you provide a library for Arduino, however other than the library itself, the T-WATCH S3 is poorly documented, there is very little information about how the device works, how to set the PMU, and things like that.

This are a list of initial questions I have:

1. At startup what is the AXP2101 configuration? All the LoRa modem, the display, and all the other things already powered up?

2. The AXP2101 i2c bus seems to be connected to pins 10 and 11 of the ESP32. I downloaded the data sheet. Please can you tell what configuration I should set in the chip in order for the T-WATCH to work? Also, if I mess with the registers, is it possible to damage the T-WATCH with wrong voltages?

Thanks

antirez, 2024-01-27:
For instance from the schematic it looks like that the display receives power from ALDO2 pin of the AXP2101. So I wonder if on power-on this pin is already setup to have the tension required or not.

antirez, 2024-01-27:
I made some progress. After reading the datasheet of AXP2101 I could not talk with the device via i2c because in the device datasheet a wrong slave address for the device is specified. The address is 0x34 instead. Now I can read the PMU status registers and will likely discover all the things I asked here. Thanks. If I will fail I'll try to seek your help again. Thank you.

lewisxhe, 2024-01-28:
For the boot sequence and peripheral power channels, they are also explained here
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/10becb4ec620a3192c8ae0aca6390436338519bd/src/LilyGoLib.cpp#L350

If you are writing axp2101, then this may be helpful to you
https://github.com/lewisxhe/XPowersLib/tree/master/Micropython/src

antirez, 2024-01-28:
Thank you @lewisxhe, I see you are the same person writing also the AXP drivers: they are cool drivers, but for the task of just making the T-WATCH to start I'm taking a simpler approach of directly addressing the few registers I need, like this (note, this is not the final code, I'm still experimenting, but the idea is to just write the few registers needed and that's it).

```python
from machine import Pin, SoftI2C
i2c = SoftI2C(sda=Pin(10), scl=Pin(11))

if False:
    # Set above to True if you want to list all the address
    # of reachable devices. In the t-watch S3 they should
    # be 25, 52, 81, 90. Corresponding (in random order) to
    # the haptic motor, RTC clock, accelerometer and the
    # AXP2101.
    print("i2c devices: ",i2c.scan())

slave_addr = 0x34 # AXP2101 i2c slave address.

# Read PMU STATUS 1
b = i2c.readfrom_mem(slave_addr,0x00,1)
pmu_status = b[0]
print("PMU status 1 at startup", bin(pmu_status))

v = i2c.readfrom_mem(slave_addr,0x15,1)
print("[before] vbus voltage limit set to", 3.88+v[0]*0.08)
# Set vbus voltage limit to 4.36v
# Register 0x15 is Input voltage limit control.
# A value of 6 means 4.36v as volts = 3.88 + value * 0.08
# This should be the default.
data = bytearray(1)
data[0] = 6
i2c.writeto_mem(slave_addr,0x15,data)

v = i2c.readfrom_mem(slave_addr,0x15,1)
print("[after] vbus voltage limit set to", 3.88+v[0]*0.08)
```

lewisxhe, 2024-01-28:
When I wrote AXP2102, I just wrote the driver, not for T-Watch. Users can customize it freely.

antirez, 2024-01-28:
@lewisxhe indeed, it makes completely sense. But in my case all I need is to power-up the device, talk with the LoRa chip with a driver I'm porting from another project, and display some text as LoRa packets are received. And I want it to be very simple to understand for newcomers.  Anyway your drivers are a good reference: I appreciate your work.

lewisxhe, 2024-01-28:
Thank you for your understanding. Most LilyGo products are maintained by me personally. I can't do more. It would be better if your MicroPython completion can be shared with us. I will place it in README for those who need it. See.Thank you for your work too

antirez, 2024-01-28:
Sure @lewisxhe, my goal is to support the T-watch S3 here: https://github.com/antirez/freakwan, so everything will be public, but I'll also create a stand-alone repository showing how to power-up the device and how to display something on the screen and so forth. The funny thing is that I'm primarily a C programmer, and yet the Arduino / ESP32 C IDE is so annoying and slow that I prefer to go with Python for this stuff.

lewisxhe, 2024-01-28:
Awesome, thank you for your excellent work. If you have any questions later, you can contact me and I will try my best to help you.

antirez, 2024-01-28:
@lewisxhe got some initial thing working: https://github.com/antirez/t-watch-s3-micropython

Not much, but the AXP2101 is configured properly (now it charges the battery and all), and the display works with the provided library. Next stop is likely implementing the sx126x driver for LoRa.

lewisxhe, 2024-01-29:
Okay, I will continue to pay attention

McFly-Muc, 2024-11-17:
@lewisxhe I'm really struggeling with LilyGo T-PCIE which they have changed from AXP192 to APX2102... where can I find a library for this chip using in Arduino? For AXP192 I used AXP202_library but this doesn't seem to support the AXP2102 chip?
Please help, thank you so much!

lewisxhe, 2024-11-18:
> @lewisxhe I'm really struggeling with LilyGo T-PCIE which they have changed from AXP192 to APX2102... where can I find a library for this chip using in Arduino? For AXP192 I used AXP202_library but this doesn't seem to support the AXP2102 chip? Please help, thank you so much!

https://github.com/lewisxhe/XPowersLib

lewisxhe, 2024-11-21:
I have confirmed that the two libraries are written in the same way, there is no difference, and the settings are correct.

axp.setChargeTargetVoltage (XPOWERS_AXP192_CHG_VOL_4V36); can set the charging voltage to 4.36V