Xinyuan-LilyGO/TTGO_TWatch_Library #229: T-Watch S3 Write timeout

SQ2CPA, 2024-02-26:
Hello. I have 6 pcs of T-Watch S3. 
Half of them works fine but other half cannot be flashed, I got error: Write timeout

[image]

jeff-burright, 2024-02-27:
I had this problem recently. To solve it, first I did the trick with the button under the battery as described in the Readme file, then I used the esp flash tool from espressif to clear the flash on the watch. After that, I was able to connect and upload. Might be worth a try for you.

lewisxhe, 2024-02-28:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3/firmware#note

closed 2024-02-28 by SQ2CPA