Xinyuan-LilyGO/TTGO_TWatch_Library #240: TTGO T-Watch 2020 V3: A fatal error occurred: Failed to connect to ESP32: No serial data received.

rolex20, 2024-05-19:
I purchased a TTGO T-Watch 2020 V3 last year and I completed the program I needed and it worked just fine.  The last time I compiled/uploaded it using Arduino IDE 2.2.1 was on December last year.  

I haven't updated Arduino IDE since but my Windows 11 has had some updates.

Problem: When I try to upload, the final step fails with error: A fatal error occurred: Failed to connect to ESP32: No serial data received.

The program compiles just fine but the upload step fails?  How to fix this problem?

My program also compiles just fine in VSCODE/PlatformIO with latest updates.

rolex20, 2024-05-19:
Forgot to mention, both Arduino-IDE and VSCODE detect my watch connected via USB/Serial to COM3, in Arduino-IDE I can get Board-Info without issues but the last uploading steps fails with: A fatal error occurred: Failed to connect to ESP32: No serial data received.

Matheuszorba, 2024-07-16:
Did you solve your problem? Im with the same error.