Xinyuan-LilyGO/TTGO_TWatch_Library #241: t-watch-s3 - keeps rebooting with "Format SPIFFS"

gabecarm21, 2024-05-24:
I am testing the T-Watch S3 with the example codes. When I tried to upload the factory firmware (I believe it’s the original firmware), it compiles and uploads correctly, but the watch keeps rebooting with the message "Format SPIFFS." In the serial monitor, the following error appears:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f9ec6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (593) SPIFFS: spiffs partition could not be found
E (622) SPIFFS: spiffs partition could not be found
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42010a03  PS      : 0x00060c30  A0      : 0x8200ff5d  A1      : 0x3fcebc40  
A2      : 0x00000000  A3      : 0xb99cc4eb  A4      : 0x3fca08d4  A5      : 0x3c15f2af  
A6      : 0x3c15f238  A7      : 0x3c15f272  A8      : 0x8202d61a  A9      : 0x3fcebc20  
A10     : 0x3fcecfb8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000022  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff1  

Backtrace: 0x42010a00:0x3fcebc40 0x4200ff5a:0x3fcebc60 0x420336a9:0x3fcebc80 0x42007f05:0x3fcebca0 0x420079a0:0x3fcebcc0 0x420039f8:0x3fcebd40 0x42049b5e:0x3fcebd70

[image]
*Arduino Settings*

[image]

[image]
*Watch screens during reboot*

If I use the partition scheme: Default 4MB, as shown in the image below, I can upload some of the examples. However, if I use the 16M Flash (3MB App/9MB FATFS), which is recommended by the library, none of the codes work; the screen goes black or the watch keeps rebooting.

[image]

I'm using Arduino IDE 2.3.2, and the esp32 library is version 2.0.9

lewisxhe, 2024-06-12:
Oh! Yes, it was a mistake, thanks for reporting it

lewisxhe, 2024-06-12:
Fixed, changed the partition to FATFS to be compatible with Arduino IDE partition table settings