Xinyuan-LilyGO/TTGO_TWatch_Library #244: example codes not compiling due to fucntion  ledcDetachPin(_pin);    ledcSetup(_channel, 1000, 8);         ledcAttachPin(_pin, _channel); on the file bl.h file

IshaSheikh, 2024-06-10:
TTGO_TWatch_Library-master\src\drive\tft\ bl.h
...
In file included from c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/TTGO.h:105,
                 from c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/LilyGoWatch.h:406,
                 from C:\Users\Lenovo\AppData\Local\Temp.arduinoIDE-unsaved2024510-17676-13wnjmw.vms4\AnalogRead\config.h:16,
                 from C:\Users\Lenovo\AppData\Local\Temp.arduinoIDE-unsaved2024510-17676-13wnjmw.vms4\AnalogRead\AnalogRead.ino:1:
c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/drive/tft/bl.h: In destructor 'virtual PWMBase::~PWMBase()':
c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/drive/tft/bl.h:18:9: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
   18 |         ledcDetachPin(_pin);
      |         ^~~
      |         ledcDetach
c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/drive/tft/bl.h: In member function 'virtual void PWMBase::begin()':
c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/drive/tft/bl.h:23:9: error: 'ledcSetup' was not declared in this scope
   23 |         ledcSetup(_channel, 1000, 8);
      |         ^~~
c:\Users\Lenovo\Documents\Arduino\libraries\TTGO_TWatch_Library\src/drive/tft/bl.h:24:9: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   24 |         ledcAttachPin(_pin, _channel);
      |         ^~~~~
      |         ledcAttach

rachanaa04, 2024-06-10:
Even I am having the same issue

Torresito99, 2024-06-11:
thats cause you had actualized  to the  3.0.0 version and as I see the code did not work for that version

lewisxhe, 2024-06-12:
Yes, the README states that you cannot use versions 3.0 and above, and it is best to use 2.0.9

rachanaa04, 2024-06-12:
Then how to make work with 3.0.0 version with ttgo t-watch v3 device ?

IshaSheikh, 2024-06-12:
> Yes, the README states that you cannot use versions 3.0 and above, and it is best to use 2.0.9

which version are you talking about..? arduino IDE version, TFT_eSPI library version... version of what? kindly clarify and I cannot find where in the readme it mentioned about version of.

lewisxhe, 2024-06-12:
[image]

IshaSheikh, 2024-11-26:
thanks for your extensive support and clarification

closed 2024-11-26 by IshaSheikh