Xinyuan-LilyGO/TTGO_TWatch_Library #253: Is it possible to receive ASK modulated message over 433 MHz?

sguerrini97, 2024-10-30:
Hello,

we are planning to buy some T-Watch S3 and we would like to pair them with "call buttons".

The buttons we already have are these and they basically send an ID over 433 MHz using ASK modulation:
[image]

We also own a T-Watch S3, 433 MHz version (K217) for our tests.
We are trying to receive the message from the buttons on our T-Watch S3 but to no avail, is this an SX1262 limitation or is this a software limitation? We tried some of the samples but none of them seems to match our setup.

Thank you
Kind regards

lewisxhe, 2024-10-31:
SX1262 does not support ASK
[image]

sguerrini97, 2024-10-31:
I see, thank you very much :)

closed 2024-10-31 by sguerrini97