Xinyuan-LilyGO/TTGO_TWatch_Library #256: Lilygo watch s3 issue

activemarkus, 2025-02-07:
We have uploaded our own firmware onto these watches.  It seems with no rhyme or reason, occasional a watch will turn itself on without me turning it on.   Is there something in the hardware making it do this? 

We also have noticed the battery being volatile. 

I opened 10 brand new watches directly from lilygo and charged them to 100%.
Turn them all on, set them all on the test table.

2 hours later, some of them are 10-15%, one st 38%, some in 50-60%c and one at 89%.

How is this possible?  Faulty batteries from lilygo?   Something drawing the battery heavily? 

Please help.  Thanks.

lewisxhe, 2025-02-07:
The AXP2101 battery management used by T-Watch-S3 is not calibrated for the battery, so the fuel gauge is inaccurate. It is more direct to get the voltage directly

activemarkus, 2025-02-07:
Ok, so how do I have this display more accurately?

Also what about my other question about it randomly powering up whenever it
wants with no reason?

activemarkus, 2025-02-07:
> Random startup? Is it in shutdown mode? Or in sleep mode?

When it is completely off ( Golding crown for 6 seconds).

Sometimes come in and it has powered on itself…
What would cause this?

Last night I even tested this.  Turned 10 watches off.  Came back into
office this memory and 3 of them were on.

lewisxhe, 2025-02-07:
Oh! This is not normal. If you long-press to turn off the phone, it shouldn't wake up. How long will it take for it to automatically turn on? Can you test it?

activemarkus, 2025-02-07:
There is no way for me to test it because it is just random sometimes.
Maybe 1-4 hours later and I notice one or 2 are on.

lewisxhe, 2025-02-07:
I mean, does it always turn on automatically? Or sometimes it does and sometimes it doesn't?

activemarkus, 2025-02-07:
Sometimes it does, sometimes it doesn’t.  Not always.  And not all of the
watches…..no reason, just random.
Also our  firmware is just lvgl and using your own library from GitHub.

activemarkus, 2025-02-07:
Also how do we get a battery meter to work properly?

And how would you optimize for battery longevity?
We are only able to get 2 hours or so right now which is way too little.
We need WiFi on for our esp32 to work and constantly receiving signals..,

lewisxhe, 2025-02-07:
There are a few problems. Contact the customer service to resend them to you.

Turning on wifi is bound to not save power. The average current consumption of turning on wifi is above 170mA.

activemarkus, 2025-02-07:
Which customer service?  Resend what to me?

Also what do you mean current consumption of turning on WiFi is above
170mA?

17mah per what?  Per hour?

I beleive the WiFi is mandatory for esp32 to work correct?

lewisxhe, 2025-02-07:
1. There are several devices with problems. Contact the dealer to resend them to you.
2. The power consumption of turning on wifi is very high. You can refer to the esp32s3 design manual.
[image]

activemarkus, 2025-02-07:
Why do you think the brand new watches are faulty?

How do we get a calibrated battery meter?

How do we optimize to get more then 2 hours?

Why do the watches discharge at different rates with the same firmware?

lewisxhe, 2025-02-07:
A normal watch cannot automatically turn on after being completely turned off, so I judged it to be a malfunction.

As I said, the fuel gauge is not calibrated, so the percentage has no reference value

activemarkus, 2025-02-07:
> How do we calibrate?  Is there a way to improve this or is it just a
> useless feature?

???

ssieb, 2025-02-07:
> Why do the watches discharge at different rates with the same firmware?

Are they?  Or is that just the calibration issue?  Have you tried running them to power off?

lewisxhe, 2025-02-08:
What parameter do you mean by discharge rate?

Have you tested the discharge current between devices with different discharge rates?

Your application requires the wifi to be turned on for a long time, and it cannot be optimized too much. And if the wifi is turned on for a long time, it is obvious that this product is not suitable for this scenario.
You need to change to a device that supports a larger battery capacity.

activemarkus, 2025-02-11:
Also, while plugged in…why do some watches just turn on?  Is it because
they are full charge?

lewisxhe, 2025-02-11:
PMU has the function of automatically turning on when USB is plugged in, and cannot be turned off

activemarkus, 2025-02-11:
Well…while plugged in, some will turn off.  Some will not allow turning off…

activemarkus, 2025-02-11:
Also, when will the 940mah version of the t watch be available for
purchase?

activemarkus, 2025-02-17:
We have multiple t watches here by lilygo.  We cannot get the lora to work.
We are on the correct frequency, and have flashed 2 watches with the
correct firmware supplied by lilygo.  They still will not communicate with
each other.

Our own firmware also does not work, even though we flash our firmware to
other lora devices and it works just fine.

Here are even 2 examples from lilygo which won’t work on our t watch s3s

examples\radio\SX1262\SX126x_Receive examples\radio\SX1262\SX126x_Transmit
These examples

lewisxhe, 2025-02-17:
Did you write directly into the example? Did you change the frequency setting? What does the serial display show?

activemarkus, 2025-02-17:
Yes, I did.  No, frequency is set correctly to watch.
Serial display?

lewisxhe, 2025-02-17:
The receive example doesn't show up on the screen but prints on the serial, can you post the example you used?

lewisxhe, 2025-02-17:
The example has been updated, you can test it

activemarkus, 2025-02-18:
Why are we perfectly capable of getting the esp32 to work on the watches
but not the Lora?  It seems like a hardware issue.

We would just continue to use esp32 except it is unreliable at times with
signals, and the battery drain is massive.

Please help!

lewisxhe, 2025-02-18:
Why can't LoRa run? Can't the above two examples run?

ESP32's wifi consumes a lot of power. This is a known problem. There is no other solution.

activemarkus, 2025-02-18:
No, they do not run on the watches!

activemarkus, 2025-02-18:
> Please run the example and then provide the serial message

They won’t run.

lewisxhe, 2025-02-18:
OK, I don't have a watch, I will send the compiled firmware to the factory for testing, and give you feedback later on whether there is any problem with the example

lewisxhe, 2025-02-18:
OK, no problem, which step went wrong?
[image]

activemarkus, 2025-02-18:
> OK, no problem, which step went wrong?
> image.png (view on web)

Yes please send me compiled firmware and I’ll test now.  That speeds things
up.

lewisxhe, 2025-02-18:
[build.zip](https://github.com/user-attachments/files/18840792/build.zip)

activemarkus, 2025-02-18:
And is this transmitter or receiver?

lewisxhe, 2025-02-18:
There are two firmwares, corresponding to sending and receiving. These are the firmwares for the two functions.  esptool flashing, address 0x0

ssieb, 2025-02-18:
> They won’t run.

What does that mean?

activemarkus, 2025-02-18:
What frequency is on these firmware?

activemarkus, 2025-02-18:
> No need to select other options, just write this file to address 0x0, no
> other options. If you use espdownloadtool, no need to select other options,
> just keep the default

Yes this file seems to communicate.  But what frequency is this on?  We
need to adjust our own transmitters to meet your watches.  I need frequency
please.z

lewisxhe, 2025-02-18:
This one uses 868M

activemarkus, 2025-02-19:
We have gotten the lore module to work. But the range seems terrible.

How much range distance should the lilygo watch be capable of?

lewisxhe, 2025-02-19:
The open distance is at least one kilometer. If there are obstacles in the middle, the distance will be lost a lot.

activemarkus, 2025-02-19:
Well, we just used our hardware with added internal antenna.   50 feet lost
connection, no signals being accepted by the watch.

Rssi - 25 with 1 foot distance.
Rssi- 60 with 10 foot distance
Rssi - 118 with 30 foot distance.

No signals at 50 feet…..

activemarkus, 2025-02-19:
are you saying 1km with Lora?  Or loRaWAN?

lewisxhe, 2025-02-19:
Of course it refers to LoRa. If there is LoRaWAN, the distance is farther.

What parameters and frequencies did you use?

CW, SF, TX POWER, these key parameters affect the distance. If you don’t understand, please check the relevant parameters of lora.

This example tells you how to set these parameters.

https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3/examples/radio/SX1262/SX126x_Settings

activemarkus, 2025-02-20:
> BW,CW, SF, TX POWER, these key parameters affect the distance. If you

Our issue now, is watch to watch is fine, bitcwatch to our own pcb is not.
If using the stock firmware on watch it is getting very poor rssi from our
pcb.  But if we try to adjust ANY parameters on the watch firmware then it
no longer receives signals…

activemarkus, 2025-02-20:
> What type of LoRa does your PCB use?

Sx1262

lewisxhe, 2025-02-20:
What parameters are used, is the LoRa at a uniform frequency, and does the hardware pre-filter match the frequency set by the software?

activemarkus, 2025-02-20:
If we use the stock firmware you provided, and our firmware on PCB, it gets
extremely poor signal. Then if we adjust your firmware or parameters even 1
thing, we get white screen upon boot up with no “waiting for transmission”
message.

On serial monitor, the watch seems fine and says it’s “searching” but
receives nothing.

lewisxhe, 2025-02-21:
Post your minimal test code .To solve the problem, please post your minimal test code. And tell us your test setup.

activemarkus, 2025-02-23:
We’ve got lots paired to the watch.  The main issue now is:
Anytime we change some parameters, the watch stops getting signals all
together from the transmitter.

Also, it will get smaller messages, but anything more than 5 character
message it will not receive?

lewisxhe, 2025-02-24:
The transmitter and receiver of LoRa must use the same parameters to receive. Do you know this? Any different parameters will be filtered out.

The maximum payload is 255. The larger the payload, the longer the transmission time. If the transmission speed is too high, some messages will definitely be lost.

activemarkus, 2025-02-24:
How our issues are different.  Seems like it takes a long time sometimes to
get response from your technical team.

Currently with Lora are only getting 3 hours of battery life which is
terrible.  Even esp32 got 3.5….
What could cause such a massive draw?

lewisxhe, 2025-02-25:
Under what circumstances was the 3-hour lifespan estimated?

Screen status? WiFi status? LoRa on? How do you control so many peripherals? Is there deep sleep? Or light sleep? Is the operating frequency reduced? What is the backlight level?

activemarkus, 2025-02-25:
Let’s start by discussing.
When I told you my issue, you said to switch over to Lora and it would
drastically improve battery life over esp32.

After $2,000, and 10 days of rebuilding, That is not the case….same UI,
same everything.

lewisxhe, 2025-02-25:
1. It is generally accepted that LoRa consumes less current than WiFi during transmission
2. What kind of optimization have you done to improve the lifespan? Just like the question I mentioned above, what have you done to save power consumption?

activemarkus, 2025-02-25:
What is the default current limit on the LoRa?

lewisxhe, 2025-02-25:
https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000Un7F/yT.fKdAr9ZAo3cJLc4F2cBdUsMftpT2vsOICP7NmvMo
[image]

activemarkus, 2025-02-25:
Spreading factor - 9
Bandwidth - 256ghz
Coding rate - 5
current limit set to 15ma

Why should only get 3 hours out of battery?

lewisxhe, 2025-02-25:
Spreading factor - 9
Bandwidth - 256ghz ???? gnh ??? 有这个参数吗 ？ ？ ？
Coding rate - 5
current limit set to 15ma   ???? 15 ma ????? 你这个极大的限制了发射功率，这个你可以设置为最大 ， 15mA 的限制估计发射功率0dBm都没有

lewisxhe, 2025-02-25:
Saving power consumption is achieved by sleeping, turning off the screen, reducing the backlight, reducing the operating frequency, and turning off unnecessary peripherals, rather than saving power by setting LoRa parameters. LoRa consumes a large current only when Tx, and maintains a current of several mA at other times