Xinyuan-LilyGO/TTGO_TWatch_Library #257: T-Watch S3 Hardware feedback

Dominaezzz, 2025-03-27:
The T-Watch S3 is fantastic and polished. Makes me actually want to use an ESP32 as a watch.
I do have some hardware feedback for the next revision if you make one. A T-Watch S3 Pro? or V2? 😄

### Use different pins for I2C buses

The device has two I2C buses. The one for the touch panel ones on GPIO39/GPIO40 and the ones for the main bus (haptic motor, rtc, bma423, etc) on GPIO10/GPIO11.
The pin choices for these two I2C buses are not optimal for low power.
The ESP32-S3 has a [ULP co-processor](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/ulp-risc-v.html) which is able to run while the main core is in deep sleep. It is able to manipulate GPIO pins, read from ADC and most importantly perform I2C operations. It's possible to use it to keep an eye on the touch screen or read data from the step counter while the ESP32-S3 stays in deep sleep. It can also wake up the ESP32 under certain conditions.
It has a restriction where it can only use GPIO0-4 for I2C. Specifically GPIO1/GPIO3 for SDA and GPIO0/GPIO2 for SCL.
Due to this restriction, the ULP can't be used to do I2C on the T-Watch S3, because different pins are used.

I propose this configuration for the next version (and all future battery powered ESP32-S3 products). This will help us save more power.

| I2C | SDA | SCL |
|------|--------|-------|
| Capacitive Touch | GPIO1 | GPIO0 |
| Main Bus | GPIO3 | GPIO2 |

Make sure to keep using GPIO4-GPIO21 for the interrupt pins and the lora chip, as those are the only pins that the ULP can use.

### Use USB C instead of USB micro-B.

This would mean one less cable to carry around and it also makes it easier to use the watch as a USB host.

### Longer wrist strap

A longer wrist strap would be nice, the watch barely fits around my wrist.

### Connect the XTAL32K pins to a 32.768 kHz clk source

GPIO15(XTAL_32k_P) and GPIO16(XTAL_32k_N) are being used for the speaker and the touch screen interrupt.

According to this [esp-idf nimble power saving example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save#typical-current-consumption-with-management-enabled), we can get 10 times the power savings if the XTAL_32k is used as the clock source. 230 uA vs 3.3 mA. More that 10 times the savings in current consumption.

According to the [schematic of the T-Watch S3](https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO_TWatch_Library/refs/heads/t-watch-s3/schematic/T_WATCH_S3.pdf) the PCF8563 has an unused RTC_CLKOUT. This RTC_CLKOUT could be connected to GPIO15(XTAL_32k_P) to get a 32.768 kHz clk source, and we'd be able to save more power.

I would be grateful if any these could be implemented in the next version. Most importantly the I2C and XTAL_32k notes.
Thanks!

lewisxhe, 2025-03-27:
Thank you for your feedback, all your feedback will be updated in the next version.

Dominaezzz, 2025-03-27:
Thanks for the quick response. I'm looking forward to the next version! ♥️

Dominaezzz, 2025-04-25:
I've added more detail [here](https://github.com/Dominaezzz/Optimal-Development-Board-Guides/blob/main/ESP32-S3.md).

lewisxhe, 2025-04-26:
Thanks for your detailed introduction. The pin mapping of the latest version of the watch can be found here.
For the old version, keep the status quo.

https://github.com/espressif/arduino-esp32/blob/master/variants/lilygo_twatch_ultra/pins_arduino.h

About more USB, this cannot be done because the size of the watch is limited.

The new version of SPI is not arranged in the best way. This cannot be changed at present, but there is very little that can be done on ULP, right? I2C may be more important for ULP.

Dominaezzz, 2025-04-26:
Yes, the ULP can do very little with SPI so it's not important. I2C is the important bit.

Looks like the SDA and SCL pins are the wrong way around on the new pin mapping so they can't be used by the ULP.
Is it too late to swap them around? To make it SDA=3 and SCL=2.

lewisxhe, 2025-04-26:
https://github.com/espressif/arduino-esp32/pull/11299
No, I already submitted an update request a few days ago

Dominaezzz, 2025-04-26:
Oh awesome!

I'm also glad to see that you've exposed the SD card detect pin too. I wish more dev boards would do this.

Dominaezzz, 2026-01-07:
Any updates on the T-Watch Ultra? I'd like to write some Rust firmware for it.

lewisxhe, 2026-01-08:
It has already been sold. 
https://lilygo.cc/products/t-watch-ultra?variant=51364248158389
However, it seems to be out of stock.