Xinyuan-LilyGO/TTGO_TWatch_Library #258: T-Watch S3: LoRa SX1262 No Packet Reception Beyond 3 Meters

ggranata89, 2025-04-11:
I have successfully programmed the T-Watch S3 to receive LoRa packets. However, I am encountering a significant range limitation. The device is unable to receive any packets when the transmitting device is located at a distance greater than approximately 3 meters.

My settings:
Frequency = 868MHz;
Bandwidth = 125KHz;
SpreadingFactor = 10;
CodingRate = 6;
SyncWord = AB;
Preamble = 12;

Could you please provide any insights or suggestions on how to troubleshoot this limited reception range? Are there any known issues with the SX1262 module on this specific device, or any specific configuration steps I might be missing?

lewisxhe, 2025-04-14:
3 meters is a very short distance, and it shouldn't be that short even with the default power.

Try changing the frequency to test. What version did you buy? 868 or 433? What frequency did you use to test?

ggranata89, 2025-04-14:
I buyed  868 version.
Thanks.

lewisxhe, 2025-04-15:
If both are SX1262, then they use the same RF front end. How many watches do you have? Only one?

lewisxhe, 2025-04-16:
Well, you said you have multiple T-Beam Supremes, I believe you should have experience.
Maybe there is a problem with the antenna of the watch, please contact the dealer, they will handle your problem, just send them this issue link.