Xinyuan-LilyGO/TTGO_TWatch_Library #259: t-watch-s3 enters bootloop every other time

LukasKuhlmay, 2025-05-05:
16:03:39.925 -> ESP-ROM:esp32s3-20210327
16:03:39.925 -> Build:Mar 27 2021
16:03:39.962 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
16:03:39.962 -> Saved PC:0x403cebfe
16:03:39.962 -> SPIWP:0xee
16:03:39.962 -> mode:DIO, clock div:1
16:03:39.962 -> load:0x3fce3808,len:0x44c
16:03:39.962 -> load:0x403c9700,len:0xbe4
16:03:39.962 -> load:0x403cc700,len:0x2a38
16:03:39.962 -> entry 0x403c98d4

When restarting the watch using the crown power knob, every other time the watch enters a bootloop shown above. 
The code which is running on the watch does not change the described behaviour. I tested with the my firmware, the factory example and the esp32 GetChipID example code.

lewisxhe, 2025-05-06:
Did you select the correct board options?
Open ArduinoIDE -> Tools
Board -> ESP32S3 Dev Module
USB CDC On Boot -> Enable ## Note that you need to change Enable to Disable when USB is not connected, so USB CDC will not prevent the board from starting
CPU Frequency -> 240MHz
USB DFU On Boot -> Disable
Flash Mode -> QIO 80MHz
Flash Size -> 16MB(128Mb)
USB Firmware MSC On Boot -> Disable
PSRAM -> OPI PSRAM
Partition Scheme -> 16M Flash(3MB APP/9.9MB FATFS)
USB Mode -> Hardware CDC and JIAG
Upload Mode -> UART0/Hardware CDC
Upload Speed -> 921600

LukasKuhlmay, 2025-05-06:
yes, i used these board options. I found that when disconnecting the battery using the switch, the watch enters the bootloop less frequent as when the battery is connected. Also when flashing the esp, the firmware does not get started right away and therefore I have to restart the watch.

lewisxhe, 2025-05-06:
After entering the download mode, you must manually restart the device, otherwise it will not exit the download mode.

Have you uploaded a simple hellworld to test whether the esp works properly?

LukasKuhlmay, 2025-05-06:
The negative battery lead broke off the board so I had to solder it back on. It seems like I created a connection which should not be there because now after redoing the soldering the problem has gone.

LukasKuhlmay, 2025-05-06:
Yes, i uploaded a simple hello world but the problem is still there. I encountered that when the esp is turned off after beeing in a  bootloop and i wait some time (around 5 minutes) until I turn the watch back on, the watch boots as intendet.

lewisxhe, 2025-05-07:
If a simple hellworld causes the USB to keep flashing, then it is suspected that the hardware is damaged. Can you take a video to illustrate the situation?

LukasKuhlmay, 2025-05-07:
https://github.com/user-attachments/assets/2a39dc9d-87d9-4d2e-a3b3-bb9111d356e0

lewisxhe, 2025-05-07:
Which version of espcore did you install? Try using version 2.0.14 to see if the problem is the same.

LukasKuhlmay, 2025-05-14:
I have been using 2.0.9 as recomended. The problem is the same when using version 2.0.14.

lewisxhe, 2025-05-15:
Finally, in order to eliminate environmental factors, write the compiled binary file for testing. After writing, you must turn off the watch and then turn it on again. If the same problem still occurs, please shoot a video and send it to the dealer for after-sales service.
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/t-watch-s3/firmware